uvic-auvic/Firmware-Power_Board_1.0 #1: Blinky LEDs

lymacasm, 2017-09-27:
**Goal:** Get the status LEDs blinking how we want them to.  
 
**Tasks:**
- Setup GPIO pins PC8 and PC9 as output pins for the LEDs.
- In the blinkyTask, toggle the green LED (PC8) at a frequency of 2Hz.
- When the board first boots up, turn on the blue LED (PC9), and keep it on for 2 seconds, before turning it off. Once this LED is turned off, the green LED should start blinking in the blinkyTask.
 
**Why do we want this?**
- The **blue** LED is used to indicate if the board has been reset. On the power board, we have the option to receive power from both the battery or the USB. This firmware will allow us to see if switching between these two power sources will reset the board. Also, it can be useful to see if the board has reset for any other reason.
- The **green** LED consistently flashing at a fixed frequency will help us to determine if the board is still alive. We can also change this frequency when flashing to ensure our changes have been loaded onto the board. If the firmware were to crash (or trap) for some reason, the LED would stop flashing, and this would give us a visual indication of the change.

anijjar, 2017-10-16:
void blinkyTask(void *dummy){
	RCC_AHBPeriphClockCmd(RCC_AHBPeriph_GPIOC, ENABLE);
	GPIOC->MODER |= 0x50000;
	GPIOC->ODR |= 0x100; // 0001 0000 0000
	vTaskDelay(2000);
	GPIOC->ODR &= ~(0x100); //clears pin 8 (0b0000 0000 0000)
	while(1){
		GPIOC->ODR ^= 0x200;
		vTaskDelay(500);
	}
}

lymacasm, 2018-02-28:
Pull request has been merged into master.

closed 2018-02-28 by lymacasm