uvic-auvic/Firmware-Power_Board_1.0 #8: Get USART working on the Power Board

lymacasm, 2018-01-31:
**Task**
Get USART working on the power board. The Power Board uses USART1 for communications. It uses an STM32F0 microcontroller. USART should somehow buffer the incoming and outgoing data, and deal with that as it goes.

lymacasm, 2018-03-05:
As per previous question, the UART pins are PA9 and PA10 (RX and TX, respectively). You can see the schematic below:

[image]

bates-andy123, 2018-03-05:
Who ever does this should look at the code in Motor_Controller_1.0 . It already does most of what you need. You can just improve on it from there

anijjar, 2018-03-05:
sounds good.

bates-andy123, 2018-05-10:
I just imported the motor controller code, issue done

closed 2018-05-10 by bates-andy123