uvic-auvic/Firmware-Power_Board_1.0 #22: Correct components on the BOM through altium

bates-andy123, 2018-05-16:
Certain componenets did not order because they were incorrectly placed in altium.

lymacasm, 2018-05-16:
The diodes for the relay were too small for the footprint, and they also didn't show up in the BOM. The connectors for 5V, 9V and 12V weren't part of the BOM but we're ordered anyways. The buck converter for the 5V5 line wasn't in stock so we're waiting for that. Those are all the part issues I am aware of. I confirmed that the only parts not ordered due to BOM issues were the relay diodes, but there could be other parts that don't match the footprint that I'm not aware of.

lymacasm, 2018-05-16:
The only solved issue is the connectors, though they still won't show up in the BOM.