uvic-auvic/Firmware-Power_Board_1.0 #24: LMV324LIPT not behaving like true rail-to-rail op-amp

poornack, 2018-05-17:
The LMV324LIPT op-amp's output reaches a maximum of 2.7V and a minimum of 45mV when supplied 3.3V and Ground. This op-amp returns values from the two battery current sensors, external pressure sensor and the water sensor. This shouldn't cause an issue when operating under normal condition so I am marking this as WONTFIX for now. We can revisit swapping this op-amp for another in the future.

lymacasm, 2018-05-17:
Should be fine. If we're worried, we can also change the value of the resistor at the output of the current monitor IC, as that resistor is used to adjust the gain. With the current resistor (time current, not electron current), the only time we'll have a 3.3V output is when we have ~120A output from the batteries, which is both the maximum battery output current, and a scenario that will likely never occur, especially since the board is fused. But, if we are worried about measuring the current in the upper threshold, the easiest thing to do would be to swap the resistor to one that will produce 2.7V at 120A.

lymacasm, 2018-05-17:
At 2.7V in the ADC, the battery current should be at approximately 98A (120A * (2.7/3.3)) which is too large anyways, so we can assume there's a problem if we see that